dmroeder/pylogix #186: Python versions compatibility to be added to README.md

xandrade, 2021-11-10:
### Preflight checks

Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [x] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [x] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [x] Have you read previous issues?

## Type of issue

- [ ] Bug
- [ ] Feature Request
- [ ] Question
- [x] Other

> Delete items that do not apply below.

## Description of issue

To add the Python versions compatibility to README.md file (3.7, 3.9, etc)

## Expected behavior

## Actual behavior

## Code

Please provide a minimal, reproducible example

## Screenshots

## Stacktrace

## Versions

Include versions to

- pylogix:
- python:
- OS:

TheFern2, 2021-11-10:
This is a good point, pylogix supports python2 and 3, but yeah I agree is not explicitly said in the README.

dmroeder, 2022-04-14:
Added a while back

closed 2022-04-14 by dmroeder