dmroeder/pylogix #191: Reading Status Bits on Micro850

dpatel2016, 2022-02-01:
### Preflight checks

Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [X] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [X] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [X] Have you read previous issues?

## Type of issue

- [ ] Bug
- [ ] Feature Request
- [X ] Question
- [ ] Other

> Delete items that do not apply below.

## Description of issue
I am trying to read the status bits in Micro850 and Pylogix does not read it. These status bits are global variables. 

## Expected behavior
I expected this to work similar to made up global variables. 

Include versions to

- pylogix: 0.7.16

dmroeder, 2022-02-02:
I don't think Micro800 supports reading those status variables in the same way that they don't support reading local variables.

dpatel2016, 2022-02-02:
One of the status bit was __SYSVA_MAJ_ERR_HALT Major Error Halted status, but if reading that is not supported, then any idea on how else to read it. I can make a variable and copy the status to it in Ladder logic, but I am sure when it faults, the program won't work so the status will not go from false to true.

dmroeder, 2022-02-02:
Just to be sure, I setup FactoryTalk to talk to a Micro820, it is not able to read the status tags either.

I think your only real option is to create a self resetting timer (its DN bit XIO in front of the timer) and read the accumulated value.  When the value stops changing, it is either faulted or no longer in run mode.

I see a lot of people spending time dealing with what happens after a fault when they should really be preventing the fault in the first place.  Sanitize values before attempting to use them indirectly, don't put negative numbers in timer presets.

dpatel2016, 2022-02-02:
I will try that! All my programs are tested before they go to implementation on the floor, but if a hard fault occurs due to I/O, then I would like to be able to see that on my IIoT dashboard. I am just trying to monitor as much info as possible including an RTC battery as well so I know when to replace it. 

Thank you.

dmroeder, 2022-02-02:
Makes sense. I wish those things weren't so limited in the way that they are. They could at least give access to the local variables

kyle-github, 2022-02-02:
Some PLCs have only one CPU and when they fault, the whole things hangs.   If the Micro8x0 series is like that, then you won't get anything after a fault.  Can you read other tags after a fault happens?

dmroeder, 2022-02-02:
Modern Rockwell PLCs still communicate when faulted, unless a type 1 code 60 /61 fault happens, which wipes the memory. I can't speak for the Micro800 though, I don't use them

closed 2022-09-10 by dmroeder