dmroeder/pylogix #198: read tag from 80 sets of plc

shuishuiz, 2022-07-15:
### Preflight checks
i work on a car manufacturing.my bodyshop has 80 sets of plc.i want to read tags to record time that how long the cycle time.
what can i do?
Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [ ] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [ ] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [ ] Have you read previous issues?

## Type of issue

- [ ] Bug
- [ ] Feature Request
- [ ] Question
- [ ] Other

> Delete items that do not apply below.

## Description of issue

## Expected behavior

## Actual behavior

## Code

Please provide a minimal, reproducible example

## Screenshots

## Stacktrace

## Versions

Include versions to

- pylogix:
- python:
- OS:

shuishuiz, 2022-07-15:
i read other people's question. i need my connection always on.my plc is guardlogix L71.

shuishuiz, 2022-07-15:
I worry about the function of write tag. It is safety?

TheFern2, 2022-07-15:
> I worry about the function of write tag. It is safety?

If you write to a tag, and you don't know what you're doing, then no, it is not safe. The safety is your responsibility, not ours.

This contradicts your initial question about reading from 80 plc's, if all you need is read then I am not sure why you're worried about the write method, but again is your responsibility.

Without some form of concurrency or parallelism is going to be really tough to collect data if is happening simultaneously in all 80 plc's, remember python is blocking by nature. If the cycle completion is different for all 80 plc's, then I would personally setup an event handler based on some bool tag or something that tells you a cycle is completed so you can then collect whatever data on that plc without constantly polling all the data on all 80 plc's.

Though a better way to handle this would be to push all completion cycle data to a central PLC, and then from there just pull data with one read.

chrisegge, 2022-07-15:
> Though a better way to handle this would be to push all completion cycle data to a central PLC, and then from there just pull data with one read.

I agree with this. I would setup prod/cons and make a UDT for the information from each PLC and come up with a naming convention that is easily duplicated but still tells you what PLC gave you the info. 
I would keep watch on the resources for the PLC, may need to split it into 2 groups if you have too much going out. I can't remember what limitations the L71 has on that. An L80 series would be better suited to connect to that many PLC's, plus whatever devices its already connected to.

> I worry about the function of write tag. It is safety?

Are you worried about writing to a Safety Tag since it is a Guard Logix? 

> I plan to open 80 connections. After the CT of each vehicle is completed, PLC will make a record, and then pylofix will collect it

I am sure you could write the code to open all 80 connections and keep them open. I just don't know if that is the best way to do it. Maybe you could write it where you cycle through each one checking for updated information.

TheFern2, 2022-07-15:
Yeah if the limitations of L71 don't allow that many connections then you have to get clever.

Another option would be to modify the plc code if you own it, to save last N cycles data.

That way it becomes an easier task for pylogix to collect data without worrying too much about concurrency.

In a forloop you connect to plc 1, get all current N cycles, store in a dictionary last cycle timestamp, print to log...

On the next connection you get all current N cycles but discard anything before last cycle timestamp, then print to log new cycles.

Then you just do that forever. I wouldn't keep 80 connections open though probably will use too much system resources. You could perhaps open 5 or 10 per thread. What have you tried so far?

dmroeder, 2022-07-15:
I agree 100% with @TheFern2

I probably wouldn't pull all of the data to 1 PLC, especially via Producer/Consumer.  You'd have to stop all 80 PLC's to accomplish that.  I'd have each machine track it's own data, use pylogix to iterate though each machine, reading the history when it is convenient.  Keep it simple.

I have no idea what it would be like to try to keep 80 connections going, that would be an interesting experiment for knowledge.  If I were to try it for fun, I'd start with a few and see the result, using data.  But in the end, I'd do what TheFern advised.

shuishuiz, 2022-07-16:
I am glad to receive all reply.
I can modify PLC logic.
Background:Six car models are produced in one station.I need CT of each model,and record it into my mysql.
I think the way to save last N cycles data suits me. @TheFern2
I have a question.PLC should know that pylogix has collected data, and then PLC clears ct. Can I send a bool to PLC through pylogix to clear CT?Or without the write function, let pylogix calculate CT and record it.Pylogix only needs to read the start timing signal of PLC.   About 200 stations.
I taught myself Python for half a year and didn't understand all the good programming ideas So ask a Daniel for advice

shuishuiz, 2022-07-16:
I am also interested in testing the results of opening 80 connections At this stage, I want to do the CT function first

shuishuiz, 2022-07-16:
The tag I wrote is not safe, but standard.
Communication is very mysterious to me. I'm worried about dislocation in the transmission process.

dmroeder, 2022-07-16:
Pylogix returns the status of Read() and Write().  Just check that .Status was "Success" with each read/write.

dmroeder, 2022-07-16:
>So ask a Daniel for advice

Who is Daniel?

shuishuiz, 2022-07-16:
the translation software is wrong.
what i want to express is to ask a highly skilled person.
now i am writing a demo to test one station...

shuishuiz, 2022-07-16:
if i want to read a udt, refer to the example of reading time?

dmroeder, 2022-07-16:
I find it easier to just make a tag list of with the individual members of the UDT, rather than trying to parse the raw data.

For example:
```python
my_tags = ["MyUDT.Member1", "MyUDT.Member2", "MyUDT.Member3"]
ret = comm.Read(my_tags)
```

shuishuiz, 2022-07-16:
[image]
still b'....'.

dmroeder, 2022-07-16:
Not sure what happened with that last comment.  Looks like you might have tried attaching a picture

shuishuiz, 2022-07-16:
from pylogix import PLC
comm = PLC()
comm.IPAddress = '10.20.26.149'
CCC = ['KK1.A','KK1.B']
ret = comm.Read('CCC')
print(ret.Value)
comm.Close()

result: b'\x02\x00\x00\x00'
KK1 is my UDT. TWO bool tags(A AND B) in KK1
CCC refer to KK1.

dmroeder, 2022-07-16:
In python, putting quotes around something makes it a string.  You defined a list of strings named CCC, but what you passed to the read method was a string CCC (because you put single quotes around it)

ret = comm.Read('CCC') should actually be ret=comm.Read(CCC)

shuishuiz, 2022-07-16:
wow! it is working.
thank you very much.!!!

dmroeder, 2022-07-16:
If you need help like this I encourage joining the discord server, the link is in the main pylogix page here on GH.

shuishuiz, 2022-07-16:
ok

closed 2023-06-17 by dmroeder