dmroeder/pylogix #207: cannot import name 'Adapter' from 'pylogix'

meenavinod, 2022-10-31:
### Preflight checks

Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [ ] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [ ] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [ ] Have you read previous issues?

## Type of issue
Import Error 
- [ ] Bug
- [ ] Feature Request
- [ ] Question
- [ ] Other

> Delete items that do not apply below.

## Description of issue
Unable to Start Pylogix Adaptor 
## Expected behavior

## Actual behavior

## Code

Please provide a minimal, reproducible example

## Screenshots

## Stacktrace

## Versions

Include versions to

- pylogix:
- python:
- OS:

TheFern2, 2022-10-31:
Pylogix does not have an Adapter class to my knowledge, on which example did you see Adapter being imported?

- What are you trying to do?

dmroeder, 2022-10-31:
Pylogix does have an Adapter class, but only in the experimental branch.  Since this is not in the main branch, you must download the project from here on GH, then switch to the experimental branch.

TheFern2, 2022-10-31:
Ah ooppss, sorry for the wrong information, wasn't aware of it until now.

TheFern2, 2022-11-01:
Did you posted a comment and accidentally deleted? I see it on the notifications but is not here.

meenavinod, 2022-11-02:
Hi,
Thanks for very much for quick responses. I tried the above suggestions and finally this issue looks fine but I am still unable to communicate with  PLC. I am still facing below issues - 
1. Unable to restore L5X file to ACD.
[image]

2. I tried to incorporate L5X changes to my test project but still no connectivity with adapter
[image]
Can you please provide the ACD file with example..?

meenavinod, 2022-11-02:
I tried to import L5X IO adapter as ENBT child module. I  made some correction to add as a remote child module of ENBT. My project IO tree have adaptor module as above. 
Please let me know if I did anything wrong..?

dmroeder, 2022-11-02:
Logix is probably being picky because I exported the module from a CompactLogix.  I'm guessing that it doesn't like importing to the ENBT.  Your image above doesn't tell me how you configured the module.  The example expects that you configured it like the image:
https://github.com/dmroeder/pylogix/blob/experimental/examples/100_simple_adapter.png

The adapter in the I/O tree needs the IP address of the machine that will be running the pylogix adapter.  And your adapter script needs to be configured for your ENBT's IP address.  If that is not working, post a screenshot of your I/O adapter configuration, you exact python script and your IP adddresses.

meenavinod, 2022-11-03:
Hi, When i tried to import L5X to my project file, its giving error of parent module 
[image]

I believe adapter cant be added as Local, so I changed the parent module properties in L5X file and import again as child of ENBT 
[image]

I used same IP address as per example and used same script to run the adapter 
[image]
[image]

Still PLC and Adapter unable to connect...

dmroeder, 2022-11-03:
As I mentioned, Logix is not letting you import the module because it is terrible.  I had the module in a CompactLogix, you have a ControlLogix, it is not smart enough to disregard that difference, unfortunately :)

Is your computer's network adapter configured for the IP Address 192.168.1.23?  Is your ENBT's IP Address 192.168.1.10?

meenavinod, 2022-11-03:
Yes, IP is same as above. Its able to detect hostname correctly ...
[image]

dmroeder, 2022-11-03:
Can you capture it via wireshark and email the result to me?

I do have a ControlLogix and ENBT, though I won't be able to test until Friday.

meenavinod, 2022-11-03:
[image]

dmroeder, 2022-11-03:
Is your script running in that instance?   All I see is list identity, which is not pylogix, it is probably RSLinx.   To minimize wireshark traffic, shut down RSLinx and LogixDesigner.  Run Wireshark, run your script.

meenavinod, 2022-11-03:
[image]

Without RS Linx and Adapter Script running

dmroeder, 2022-11-04:
Here is the problem I'm having.  You show the screenshot of your generic module's Module Info tab, it is reporting a Rockwell Software Communication Adapter.  You are telling me that your computers IP address is 192.168.1.23 and that generic module is configured for 192.168.1.23, which looks to be the case and that would be correct. 

There's something up with your screenshot though.  When you open the Module Info of your generic module, it is sending out a List Identity request to the address of 192.168.1.23.  Some Rockwell device is responding.  If it is your computer that is configured for 192.168.1.23, and it should be based on your script, nothing should respond.  The only thing that should be at .23 is your computer, which would not respond to a List Identity request.

dmroeder, 2022-11-04:
Wait, can you go to the Network and Sharing Center and check if your adapter is set to "public network"?

I had trouble with this, the firewall was blocking the connect request from the PLC.

meenavinod, 2022-11-05:
I believe its reporting a Rockwell Software Communication Adapter because of RS Linx. If I change the IP address of my computer its unable to find module properties -
[image]

My network and sharing center already was in Public network but Firewall Defender was on. After disabling it After disabling the Firewall defender for Public network -
[image]

dmroeder, 2022-11-05:
Okay, you are getting close.  I don't think RSLinx would cause what you were seeing but maybe RSlinx Enterprise, I can't say for sure.  The timeout is what I would expect because the pylogix adapter hasn't implemented the list identity response.

You may have some more firewall work to do.  I think there should be a register session, which I don't see.  After both the register session and forward open, the connection switches to UDP, port 2222.

meenavinod, 2022-11-05:
I Think its registering the session but Forward Open connection failed...

I have disabled firewall completely .

[image]

dmroeder, 2022-11-05:
Can you email me that wireshark packet capture?

meenavinod, 2022-11-07:
Hi, You are right, I didnt noticed any change with Adapter Script running or not in Wireshark. Can you please share your mail address. my email is meenavinod017@yahoo.com.
[image]
 
[image]

[image]

But the Pylogix eip working as expected with same setup. 
[image]

lbhopper, 2024-03-18:
have you tried disabling keys in your Adapter Module? ControlLogix modules can return with errors if keying is set to exact or compatible:

https://literature.rockwellautomation.com/idc/groups/literature/documents/at/logix-at001_-en-p.pdf

nucleonautomation, 2026-01-16:
I built something for Ethernet Ip adapter, the adapter is in form of binaries and can be used for R&D and lab projects, or for fun 🤩 

Github: https://github.com/nucleonautomation/EnIP-Adapter

Webpage: https://nucleonautomation.com/service/enip-adapter

It has all services and classes inbuilt and a very minimal API for python.