dmroeder/pylogix #209: "Forward open failed" when trying to read tags from Berghof PLC

brunifrancesco, 2022-12-17:
### Preflight checks

Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [x] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [x] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [x] Have you read previous issues?

## Type of issue

- [ ] Bug
- [ ] Feature Request
- [x] Question
- [ ] Other

## Description of issue
This is not a real issue. This is more a question. I'm new in this field.
I'm unable to read any tags but I'm **able to discover devices** given the IP. 
I always got a **Forward open failed** error.

PLC is Berghof brand. 

## Expected behavior

I'm expecting to read them by name but I'm not sure this feature is supported in my PLC.
I check the names as in the screenshot attached. They're correct.
Slot is 0, checked in the machine conf.

## Actual behavior

I got always **Forward open failed** error for each tag.

## Code

      for index,tag in enumerate(tags.split("\n")):
          with PLC() as comm:
              comm.IPAddress = '192.168.1.15'
              ret = comm.Read(tag)
              print(ret.TagName, ret.Value, ret.Status)

That's the sample output

    Recipe_Name None Forward open failed

where *Recipe_Name* is a tag name

## Screenshots

[image]

## Versions

Include versions to

- pylogix: 0.8.6
- python: 3.10.6
- OS: Ubuntu 22.04

Any idea why this is happening? Can I try to read using memory addresses instead of names?

Forgive me If I wrote some errors, I'm newbie here.
Best, FB

dmroeder, 2022-12-17:
This is for Rockwell PLC's only.

closed 2022-12-17 by dmroeder

brunifrancesco, 2022-12-17:
Oh, that's interesting. 
Any suggestion to make it working with Berghof? 
Any other library you can think of?

dmroeder, 2022-12-17:
Unfortunately, I don't know anything about that brand or module.

brunifrancesco, 2022-12-17:
Ok, it would be too much complex try to write a simple driver? I can see it by myself but I need your help for sure.
Actually your discover function works, so I deduce that part of your code is still usable.

dmroeder, 2022-12-17:
There is a minimum amount of features a that an Ethernet I/P compliant device must support, List Identity (Discover) is one of them.  So every Ethernet I/P device will respond to that.  Communicating with it in the specific way that it is expecting is a very different thing.

My best advice is get a sharktap or a port mirroring switch and wireshark the connection and traffic.  If your next post is going to be "I don't know how to use wireshark", you have a lot of work ahead of you to write a "simple" driver.  Knowing now to use wireshark is a must.  And don't wireshark what pylogix is trying to do, wireshark a device that is currently successfully communicating with that device.

brunifrancesco, 2022-12-17:
You've been crystal. I've used Wireshark in the past, I know it.
Thanks for your time.

dmroeder, 2022-12-17:
If you find anything interesting in wireshark, you can email me.  My address is in the setup file.

brunifrancesco, 2022-12-17:
Thanks for the support!Francesco Bruni