dmroeder/pylogix #216: .GetDeviceProperties always returns DeviceType of "None"

CLau72, 2023-02-07:
When using the GetDeviceProperties method, the DeviceType is always showing as "None" despite the hex of the DeviceID value matching one of the values in the devices dictionary.

Issue looks like it's here:
https://github.com/dmroeder/pylogix/blob/20d5792f59ed9fdd7238e38b930a6a66eb8ef38e/pylogix/lgx_device.py#L113

The value is being written to resp.Device instead of resp.DeviceType

dmroeder, 2023-02-07:
Oof, you are correct, that should have been DeviceType not Device.

dmroeder, 2023-02-07:
You're more than welcome to submit a pull request if you want credit. Otherwise I'll fix it.

CLau72, 2023-02-07:
I'll go ahead and get it submitted in the next few minutes here. Haven't done a PR before so wanted to be sure I dropped this here first 👍

TheFern2, 2023-02-07:
> I'll go ahead and get it submitted in the next few minutes here. Haven't done a PR before so wanted to be sure I dropped this here first 👍

The simplest way is to fork it, and do the edit on the browser. When you do a change a pull request should appear on this repo or your fork. Then you can submit from there.

dmroeder, 2023-02-07:
Fixed in PR #217

closed 2023-02-07 by dmroeder