dmroeder/pylogix #221: Issues connecting to emulate

sethoon88, 2023-03-23:
Still having trouble with connecting to an emulator. connection size set to 504 and ip address set to the PC.  Am i missing something else?  thanks!
[image]

evaldes2015, 2023-03-23:
I think I tried this some time ago and couldn't get it to work

dmroeder, 2023-03-23:
Open RSLinx Enterprise/FactoryTalk Linx, right click on the Ethernet driver, Properties > Advnaced.  Check the box "Listen on Ethernet I/P encapsulation ports

dhunt84971, 2023-03-23:
Whoa!  Okay.  Learned something new, today.  Removing my misleading answer.

dmroeder, 2023-03-23:
@dhunt84971, no worries. there was a point where I was confidant that you could not communicate with emulate in this way.  Even after I tested it and it worked for me, it still didn't work for others.  I later learned about that configuration.

closed 2023-03-23 by sethoon88

DasDuneTech, 2023-09-02:
Hi, I am not using Factory Talk, only RSLinx Classic Gateway and Emulate 5000, everything works fine with Logix Designer 5000 btw. I get : Unknown error [WinError 10061]... with pyLogix.  Is there any setting I can do on RSlinx ?   see below a screen shot on the advanced tab in RSLinx. Thanks!
[image]

dmroeder, 2023-09-02:
I'm only aware of this option being available with FTL.  So in order to do this, you will probably have to install FTL.  You wouldn't be "using" FTL to communicate, it just acts as a bridge, accepting the connection and forwarding it to Emulate AFAIK.