dmroeder/pylogix #222: micro850 connection

fernandoamorim1703, 2023-04-18:
A question, I'm trying to use node-red's ethernet/ip node, I made some changes to the library because of the multiple service.

But I am facing connection problem. The connection is not maintained, the PLC always sends an FYN flag or even an RST.

Could you tell me why?

dmroeder, 2023-04-18:
Why are you opening an issue on my project instead of node red's EIP module?

fernandoamorim1703, 2023-04-18:
I wanted to better understand the communication with the micro850

closed 2023-04-18 by fernandoamorim1703

dmroeder, 2023-04-18:
I posted packets below of wireshark packets with the difference between CLX and Micro800.

https://www.plctalk.net/qanda/showpost.php?p=930765&postcount=28

dmroeder, 2023-04-18:
I'd pay close attention to the forward open.  Micro800 doesn't use the port segment, which is what those packets are showing.  I think I showed more of that if you read further in that thread.