dmroeder/pylogix #224: Return data gets jumbled up on a multi write when the tags don't exist.

Destination2Unknown, 2023-05-08:
## Code

    from pylogix import PLC

    with PLC() as comm:
        comm.IPAddress = "192.168.1.9"

        write_data = [("tag1", 100), ("tag2", 6.45), ("tag3", True)]

        # write the values
        ret = comm.Write(write_data)

        # print the status of the writes
        for r in ret:
            print(r.TagName, r.Status)

Prints:
```
tag2 Unknown error timed out
6.45 Unknown error timed out
```

## Versions

- pylogix: 0.8.11
- python: 3.11
- OS: Windows

Destination2Unknown, 2023-05-09:
@dmroeder Sweet!

Destination2Unknown, 2023-05-09:
I think this was the issue:

https://github.com/dmroeder/pylogix/blob/95d6bc9413cb5d7f54eb231a802dd9f2d3950e08/pylogix/eip.py#L435-L436

It should be something like:

    if not conn[0]:
        return [Response(t[0], None, conn[1]) for t in tags]

dmroeder, 2023-05-11:
You mind testing the bugfix/multi-write branch when you get time?  It works for me, though things weren't quite failing in the same way for me.

Destination2Unknown, 2023-05-15:
This fairly niche example isn't quite right either...

```
write_data = [("myDint", 100), ("myString", "Hello"), ("myDint.1", True)]

# write the values
ret = comm.Write(write_data)

# print the status of the writes
for r in ret:
    print(r.TagName, r.Value, r.Status)
```
prints:
```
myDint 100 Success
myString Hello Success
myDint.1 (2, -1) Success
```

dmroeder, 2023-05-16:
Yeah, that is weird.  The latest should fix it.

Destination2Unknown, 2023-05-16:
> Yeah, that is weird. The latest should fix it.

This case is still incorrect:
```
from pylogix import PLC

with PLC() as comm:
    comm.IPAddress = "192.168.133.199" #incorrect IP address

    write_data = [("tag1", 100), ("tag2", 6.45), ("tag3", True)]

    # write the values
    ret = comm.Write(write_data)

    # print the status of the writes
    for r in ret:
        print(r.TagName, r.Status)
```

prints:
```
tag2 Unknown error timed out
6.45 Unknown error timed out
```

dmroeder, 2023-05-16:
I see, somehow, I missed that comment originally

dmroeder, 2023-05-16:
Okay, that one should be fixed on that branch too.

I considered whether to return the value that was written or the None.  I went with value intended to write because that is what would be returned on a successful write.  Though I could be convinced otherwise.

Destination2Unknown, 2023-05-16:
> Okay, that one should be fixed on that branch too.
> 
> I considered whether to return the value that was written or the None. I went with value intended to write because that is what would be returned on a successful write. Though I could be convinced otherwise.

I strongly favour returning None because it would make it crystal clear that there was an issue.
Otherwise, it could lead to confusion when a write value is returned despite not being sent successfully.

Thank you for your assistance. 
This library is exceptional, and the support provided is truly outstanding!

dmroeder, 2023-05-16:
Sold!  I'll return None.  I appreciate your feedback and testing.

Destination2Unknown, 2023-05-16:
Looks good to me, thanks again!

commit f511dc6d (2023-05-15): Fixed multi-write when some tags don't exist

This reverts a previous commit to deal with data types that were assinged
a value of 0.  This will have to be investigated further.

Fixes #224
closed 2023-05-16 by Destination2Unknown