dmroeder/pylogix #229: Is it possible to inhibit a periodic task?

ramezanifar, 2023-06-21:
In RSLogix a task type can be Continuous, Periodic, Event. There is also a checkbox to inhibit a task.
Does anyone know if we can enable/disable the inhibit setting via pylogix? This is useful when running automated tests to disable the PLC logic to manipulate the tags from pylogix and avoid PLC logix stumping on the injected values.
Thank you in advance

TheFern2, 2023-06-23:
This is not possible with pylogix. I recommend looking into autoit scripting, is one way to automate it through the Studio5K gui.

dmroeder, 2023-06-23:
Add SSV's to inhibit the modules with a bool of choice ahead of them.  Then write to that bool with pylogix.  I think that is about all you will be able to do.

closed 2023-09-02 by dmroeder