dmroeder/pylogix #230: pylogix - returning a tag list as a row of values

wksriaanc, 2023-07-06:
Hi,

Please see function run_script() below. print r.value will print the relevant 4 tag integer values - 

50
10
25
22

By returning the r.value, I'm expecting to see the result of all 4 values in print(y) outside of the function. print(y) however only returns the last value i.e., 22. Can you possibly tell me why this might be happening? Thank you.

*** 

def run_script():    

	with PLC() as comm:

	comm.IPAddress = 'xx.xx.xx.xx'
        ret = comm.Read(list)

        for r in ret:
            print(r.value)

        return r.value
    

while True:
    y = run_script()
    print(y)

TheFern2, 2023-07-06:
In your example `ret` is a list of Response objects. So, you'll need to return `ret`, then iterate again outside the run_script function just like you did inside the function.

```python
def run_script():
    with PLC() as comm:
        comm.IPAddress = 'xx.xx.xx.xx'
        ret = comm.Read(list)

    for r in ret:
        print(r.Value)

    return ret

while True:
    y = run_script()
    for r in y:
        print(r.Value)
```

Alternatively, you can create a new integer list:

```python
def run_script():
    int_ret = [] 
    with PLC() as comm:
        comm.IPAddress = 'xx.xx.xx.xx'
        ret = comm.Read(list)

    for r in ret:
        print(r.Value)
        int_ret.append(r.Value)

    return int_ret

while True:
    y = run_script()
    for r in y:
        print(r.Value)
```

Also note that this library response object is upper case Value, not value. You might be using the wrong package. What's your pylogix version?

wksriaanc, 2023-07-06:
Fernando, my pylogix version is 0.8.14. i also have pylogix2 installed which is version 0.5.3. Assume I'm using pylogix and not pylogix2

TheFern2, 2023-07-06:
> Fernando, my pylogix version is 0.8.14. i also have pylogix2 installed which is version 0.5.3. Assume I'm using pylogix and not pylogix2

if you are running your script and calling lower case `value` then you're running pylogix2 which we are not affiliated with nor support. I suggest you uninstall pylogix2 it was a rogue fork.

Also forgot to mention you might want to add a sleep in your while loop to prevent high cpu usage.

wksriaanc, 2023-07-06:
Hi, Fernando. Thank you :) I've uninstalled pylogix2 and added a 5 sec sleep. Just one last thing - r.value as mentioned by yourself is a list of response objects. How do I store this list as a row of integer values e.g. [50, 10, 25, 22].

wksriaanc, 2023-07-06:
I might use the append method.

wksriaanc, 2023-07-06:
Fernando, the append function seems to work. Thank you again for getting me on track with my issue :)

closed 2023-07-06 by wksriaanc

TheFern2, 2023-07-06:
> Fernando, the append function seems to work. Thank you again for getting me on track with my issue :) 

Glad to be able to help.