dmroeder/pylogix #231: PLC REAL values

wksriaanc, 2023-07-06:
Hi,

When a REAL value is used in the PLC e.g., 69.8, pylogix returns 69.80000305175781. Can this be changed in the code? Thank you.

TheFern2, 2023-07-06:
Hi there, the value is exactly how is stored in the plc and how floating values are handled in general in programming. The plc software just shows you 1 or 2 points of precision.

Have a look at this tutorial, I would probably use round or f string under the second example.

https://www.geeksforgeeks.org/precision-handling-python/

TheFern2, 2023-07-06:
Here is a bit of an explanation on floating storage for 32bit cpu just as a general explanation why floating values have long precision. In short they're stored as binary numbers.

https://stackoverflow.com/questions/7644699/how-are-floating-point-numbers-stored-in-memory

Another from python docs.

https://docs.python.org/3/tutorial/floatingpoint.html

> It’s important to realize that this is, in a real sense, an illusion: you’re simply rounding the display of the true machine value.

wksriaanc, 2023-07-06:
> Hi there, the value is exactly how is stored in the plc and how floating values are handled in general in programming/cpus. The plc software just shows you 1 or 2 points of precision.
> 
> Have a look at this tutorial, I would probably use round or f string under the second example.
> 
> https://www.geeksforgeeks.org/precision-handling-python/

Hi, Fernando. I thought this might be the case. Thank you for the tutorial - I'll have a look.

wksriaanc, 2023-07-06:
> Here is a bit of an explanation on floating storage for 32bit cpu just as a general explanation why floating values have long precision. In short they're stored as binary numbers. [...]

Thank you again, Fernando. Agree with your observation - I've got no intention to round the floating values since they're as mentioned by yourself, the true value :)

closed 2023-07-06 by wksriaanc

TheFern2, 2023-07-07:
> Thank you again, Fernando. Agree with your observation - I've got no intention to round the floating values since they're as mentioned by yourself, the true value :)

You're welcome. Glad to be able to help!