dmroeder/pylogix #235: Pylogix - compare a previously acquired tag list with the current one

horus61, 2023-08-23:
Good morning everyone. I state that I am new to python programming and I would like to ask you for advice on how to compare a previously acquired tag list with the current one to check for any changes in the values. Thanks for your availability.

I used this code

def run_script():
    int_ret = [] 
    with PLC() as comm:
        comm.IPAddress = 'xx.xx.xx.xx'
        ret = comm.Read(list)

    for r in ret:
        print(r.Value)
        int_ret.append(r.Value)

    return int_ret

while True:
    y = run_script()
    print(y)

TheFern2, 2023-08-23:
Hi @horus61 I've formatted your code properly. A few comments, don't use `list` in your code as list is a python keyword, use something like `my_list` or better yet name it accordingly like `sensors` or `temp_sensors` note the plural. Also don't forget to add a sleep in the while loop or your cpu will go 100% on usage.

Here's some history on a similar topic https://github.com/dmroeder/pylogix/issues/230
How I would solve it:
```python
prev_ints = run_script()
while True:
    y = run_script()
    print(y)

    # compare lists here
    # if different, update prev_ints
    #   do something
    # else if equal, don't do anything

    sleep(3)
```

Luckily for us we have the Zen of Python https://peps.python.org/pep-0020/
```
There should be one-- and preferably only one --obvious way to do it.
```
And we have 30,000 ways to do a list comparison lol https://www.digitalocean.com/community/tutorials/how-to-compare-two-lists-in-python

I would probably use list comprehension for your use case since we do care about the order of the items in the list.

```python
l1 = [10, 20, 30, 40, 50]
l2 = [50, 75, 30, 20, 40]

res = [x for x in l1 + l2 if x not in l1 or x not in l2]

print(res)
if not res:
    print("Lists l1 and l2 are equal")
else:
    print("Lists l1 and l2 are not equal")
```

horus61, 2023-08-25:
Thank you Fernando for your help and I bother you again for further advice. I would like to identify the elements of the list that have changed since the previous reading according to their index because I would like to create an alarm report.
Thanks for your cooperation, greetings from Italy. Nino
Sorry for my bad English.

TheFern2, 2023-08-25:
You're welcome, your English is great! If you take the previous example that compares each value, you just have to create a new list, no need to get the index. Take a look below.

```python
# Initialization vars before loop is started
prev_ints = run_script()
did_change = False
changed_items = []  # <- New list
while True:
    y = run_script()
    print(y)
    # Loop through all values and compare them
    for val_y, val_p in zip(y, prev_ints):
        # Compare new value with prev value
        if val_y != val_p:
            changed_items.append(val_y)  # <- Add item to list
            print(f"The values {val_y} {val_p} aren't equal")
            did_change = True

    if did_change:
        prev_ints = y
        did_change = False
        # send report logic here  # <- New line

     sleep(3)
```

horus61, 2023-08-25:
HI. Thanks for the advice you give me. I wanted to show you the raspberry pi application that I would like to create to monitor the variables on the plc to integrate an alarm message with the sending of sms. This is an example with a variable simulator. I'm a newbie and it certainly doesn't look nice and you would do it more professionally.
But it represents what I would like to do. I thank you and greet you.

[P2TestMainFern.txt](https://github.com/dmroeder/pylogix/files/12442337/P2TestMainFern.txt)

TheFern2, 2023-08-27:
@horus61 I recommend you check out our [discord](https://discord.gg/tw8E9EAAnf), someone there might be able to further guide you. The best advice I can give you is to start small. Build each step of what you are trying to accomplish one by one on their own and then mash them together in the end. Also if you seek help in the discord your might want to translate function names and variables to English, otherwise is hard to follow the code.

For example:
- Read plc tags print them to the console
- Compare tags in a while loop
- Create fake alarm message to sms
- put all 3 programs together for end result

closed 2025-01-10 by dmroeder