dmroeder/pylogix #236: Discover does not work while running pylogix in docker

ranwang0417, 2023-09-11:
Hello

I am trying to run pylogix inside a docker container. The Read and GetTagList works fine but the Discover function cannot find any devices. I feel it relates to  #93 so I tried to use static IPv4 address for my computer (both Ethernet (Default Switch) and WIFI) and changed the subnet mask from 255.255.254.0 to 255.255.0.0 but still no luck. The response has an empty list as the Value and Success as the Status. If I run the same code but not in docker, the Discover function works fine. I was wondering if there is anything else I could try? Any help is greatly appreciated. Thanks in advance!

Here is my IP configuration:
[image]

Here are the package versions:
- python: 3.10.4
- pylogix: 0.8.3
- OS: Win 10 Enterprise, Linux docker container
- Docker base image: continuumio/miniconda3

TheFern2, 2023-09-12:
Hey there, does this also happen with a host adapter?

https://docs.docker.com/network/network-tutorial-host/

dmroeder, 2023-09-12:
I don't know much about docker but if it doesn't have direct access to a network adapter, like bridged mode in a VM, then it is likely the broadcast isn't making it out of the container.

kyle-github, 2023-09-12:
IIRC, discovery uses UDP broadcast.   That might be blocked from a Docker container.

TheFern2, 2023-09-12:
host mode is the equivalent to a vm bridge mode, that should solve the issue.

ranwang0417, 2023-09-12:
Hi @TheFern2 @dmroeder @kyle-github , thank you so much for your replies!

I was trying to create custom network for my container, but it seems that the host option is only for Linux host and transparent network is only for windows containers. I am looking into two directions now. One is to change from Linux container to Windows container and the other is to figure out how to send UDP broadcast from a docker container. I'll reply here if I could find anything.

TheFern2, 2023-09-12:
A bit off topic, but why do you need to run pylogix inside a container? pylogix requires zero deps aside python, so running in a container is imo an unnecessary overhead. Use venv or poetry if you don't want to install python on the system wide libraries, and use nssm to call your python script.

ranwang0417, 2023-09-12:
This is mainly for deployment purposes. We are developing a web-based software framework that includes the frontend, backend and database. All the components are in a docker compose network for easy deployment and they are managed by nginx. I was trying to add pylogix to the Django backend app so that we can communicate with RA PLCs. Another way around is to use the common gateway platform which claims to work in docker but has other issues that we need to resolve.

ranwang0417, 2023-09-12:
I did not find a working solution to broadcast from the container to the local network for my setup (windows host and linux container). Thank for all your help anyway!

closed 2023-09-12 by ranwang0417

TheFern2, 2023-09-13:
Yeah if docker is out of the question for discovery of devices. The next best solution is as I mentioned before, with nssm, creating either a script or even an exe for a simple api **/api/get_devices** and you call that endpoint from your docker container. Would you probably be better to write this exe in another language for a smallest binary.

ranwang0417, 2023-09-14:
Thanks for the suggestion! I'll give it a try. I really like the discover function as it could simplify the workflow and make it easier for the user.