dmroeder/pylogix #237: Invalid Continuation Byte

ifTaylor, 2023-09-27:
I received this traceback when an existing string value in the plc, not micro, contained two sentences and I attempted to write over it.
The error went away when I changed the string to  "Test" in the plc controller tags.
Although, that was the observed state, I'm having trouble replicating.

How is this so?
  
  File "\plc.py", line 63, in write_tags
    ret = self.PLC.Write(write_data) 
  File "\pylogix\eip.py", line 135, in Write
    return self._batch_write(tag)
  File "\pylogix\eip.py", line 448, in _batch_write
    self._get_unknown_types(new_tags)
  File "\pylogix\eip.py", line 1381, in _get_unknown_types
    result.extend(self._multi_read(unk_tags[len(result):], True))
  File "\pylogix\eip.py", line 425, in _multi_read
    return self._parse_multi_read(tags_effective, ret_data)
  File "\pylogix\eip.py", line 1496, in _parse_multi_read
    value = str(s.decode(self.StringEncoding))
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xcc in position 84: invalid continuation byte

dmroeder, 2023-09-27:
Is this a standard STRING type, or a custom string?  How many characters were you trying to write?

ifTaylor, 2023-10-12:
Sorry for the delayed response. More on this.

With the 4-byte unsigned integer format specifier '<I', I am able to write any length of string as long as the character groups are short enough. For example I was able to write the word "short " 385 times. The PLC truncated it to "short short short short short short short short short short short short short shor".

There were two cases that caused an error state.
1.
When I took out the space between one of the shorts, "shortshort short short.....", I received...

  File "C:\Python38-32\lib\site-packages\pylogix\eip.py", line 1571, in _make_standard_string
    temp = pack('<I', len(string)).decode(self.StringEncoding)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe2 in position 0: invalid continuation byte

2. (I began the thread here)
When I restarted my application which reads with the single-byte unsigned value format specifier '<B'. 

  File "C:\Python38-32\lib\site-packages\pylogix\eip.py", line 1496, in _parse_multi_read
    value = str(s.decode(self.StringEncoding))
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xcc in position 84: invalid continuation byte

Relating to case 1. is when I wrote "shortshortshortshortshortshortshortshortshortshortshortshortshortshortshortshortshortshortshortshortshortshortshortshortshorts". Then restarted my application and could not read. (adding an "h" makes the write fail)

Is the controller able to hold the full value? I would have thought the data would have been truncated and lost, so the format is valid when it's time to read. 
 
Moving forward I will just limit to the 82 characters studio5000 indicates.

dmroeder, 2023-10-12:
No problem.

So you were trying to write more characters than actually fit in the string?  Obviously that won't work, but I probably shouldn't let you try.  Thanks for the information, I'll do some testing.

ifTaylor, 2023-10-12:
All interesting stuff. I'm just wondering how the plc is holding onto the bad string and if it's only during runtime of the plc.

dmroeder, 2023-10-25:
Fixed by @ifTaylor

closed 2023-10-25 by dmroeder