dmroeder/pylogix #238: Abnormal code termination.

horus61, 2023-10-09:
Good morning everyone. I'm using pylogix and it often crashes with the following error. Can anyone help me solve the problem? Thank you.

Traceback (most recent call last):
  File "/home/nino/TorrreP2/TorreP2Main_Beta.py", line 118, in <module>
    dati_adesso = leggi_tags()
  File "/home/nino/TorrreP2/TorreP2Main_Beta.py", line 99, in leggi_tags
    ret = comm.Read(tags)
  File "/home/nino/.local/lib/python3.9/site-packages/pylogix/eip.py", line 120, in Read
    return self._batch_read(tag)
  File "/home/nino/.local/lib/python3.9/site-packages/pylogix/eip.py", line 279, in _batch_read
    result.extend(self._multi_read(tags[len(result):], False))
  File "/home/nino/.local/lib/python3.9/site-packages/pylogix/eip.py", line 425, in _multi_read
    return self._parse_multi_read(tags_effective, ret_data)
  File "/home/nino/.local/lib/python3.9/site-packages/pylogix/eip.py", line 1474, in _parse_multi_read
    status = unpack_from('<b', stripped, offset + 2)[0]
struct.error: unpack_from requires a buffer of at least 207 bytes for unpacking 1 bytes at offset 206 (actual buffer size is 176)

dmroeder, 2023-10-09:
Can you post a simple example that causes this issue?  It will probably be important to know the tags and their data types.

horus61, 2023-10-10:
Good morning everyone. This is part of the code. Perhaps the problem could be a time out on the network. You could create code that checks the connection and restarts automatically the program?
Thank you.

#from __future__ import print_function
import gammu
import sys
import logging
import time
#*********************************************************************************************************************
from pylogix import PLC

from datetime import datetime, timedelta
from threading import Timer 
from apscheduler.schedulers.background import BackgroundScheduler
from apscheduler.triggers.cron import CronTrigger
#*********************************************************************************************************************
# Argomenti per invio SMS ********************************************************************************************
recipients = {'******','*******','******','******','*******'} # numeri telefonici cellulari tecnici **
SMSTesto = []   # testo del SMS **************************************************************************************
#*********************************************************************************************************************

# Funzione per la pulizia dei messaggi SMS da caratteri indesiderati *************************************************
def ProcessaMessaggio(Messaggio):
    specialChars = "'[]" 
    for specialChar in specialChars:
        Messaggio = Messaggio.replace(specialChar,"")
        #Messaggio = Messaggio.replace(" ", "") 
        #print(Messaggio)  
        
        Messaggio = Messaggio.replace(",",'\n')
        #Messaggio = Messaggio.lstrip()
    return Messaggio
#---------------------------------------------------------------------------------------------------------------------------------- 

# Funzione che invia i messaggi SMS ***********************************************************************************************
def send_sms(*args):
    sm = gammu.StateMachine()
    sm.ReadConfig()
    sm.Init()

    for number in recipients:
        smsinfo = {
            'Class': -1,
            'Unicode': True,
            'Entries': [
                {
                    'ID': 'ConcatenatedTextLong',
                    'Buffer': SMSTesto
                }
            ]}
        encoded = gammu.EncodeSMS(smsinfo)
        try:
            for message in encoded:
                message['SMSC'] = {'Location': 1}
                message['Number'] = number
                message['Coding'] = 'Unicode_No_Compression'
                sm.SendSMS(message)
            logging.info('sent to ' + number)
        except Exception as e:
            logging.error(e)

#-----------------------------------------------------------------------------------------------------------------------------------------

# Schedulatore in background *************************************************************************************************************
scheduler = BackgroundScheduler()
scheduler.start()
scheduler.add_job(send_sms, args=(recipients,SMSTesto), trigger='cron',hour=9, minute=0) # Status torre P2 con SMS alle 8:00

# ----------------------------------------------------------------------------------------------------------------------------------------

[... 84 lines cut ...]
    MesLiv = []
    AlMesLiv = []

    if (LivTorreP2_OK) == True :
        MesLiv.append ("Livello "+ str(LivTorreP2/100) + " mt" + " \U0001F603")
        

        if (LivTorreP2/100) > 4.20 and (LivTorreP2_OK) == True:
            AlMesLiv.append ("Livello "+ str(LivTorreP2/100) + " mt" + " alto \U00002B06") 

        elif (LivTorreP2/100) < 2.30 and (LivTorreP2_OK) == True:
            AlMesLiv.append ("Livello "+ str(LivTorreP2/100) + " mt" + " basso \U00002B07")

    else:
        MesLiv.append("Livello " + "***" + " mt" + " \U0001F621")
        
    
    
#------------------------------------------------------------------------------------------------------------------------------------------
# Eleborazione del messaggio di testo da inviare tramite  SMS ********************************************************************************************************
    BuonGiorno = " Buon giorno da Torre "'\n'"Questo è il mio stato"'\n'
    SMSTesto = TimeStampAdesso + '\n' + BuonGiorno.lstrip() + ProcessaMessaggio(str(MesLiv ))
    print (SMSTesto)

    time.sleep(60)

TheFern2, 2023-10-10:
For future reference please provide an [mre](https://stackoverflow.com/help/minimal-reproducible-example).

For your case it would be one script with one Read call, in a while loop and a sleep of N seconds. We don't need to know about sms and other stuff going on. 

Also ensure that you are using code tags when posting so that is formatted properly.

What are the data types for the tags giving you the exception? 

Does the same exception happens when doing individual reads?

When opening an issue there's a template for some questions and you didn't provide those.

OS type and version?
Pylogix version?
Python version?
PLC model?

horus61, 2023-10-12:
#from future import print_function
#import gammu
import sys
import logging
import time
#*********************************************************************************************************************
from pylogix import PLC

from datetime import datetime, timedelta
from threading import Timer

#----------------------------------------------------------------------------------------------------------------------------------------
#Timestamp ******************************************************************************************************************************
def Data_Ora():
    DataOraReport = []
    timestamp = time.time()
    date_time = datetime.fromtimestamp(timestamp)
    str_date_time = date_time.strftime("%d/%m/%Y, %H:%M:%S")
    DataOraReport = (str_date_time)
    return DataOraReport

#----------------------------------------------------------------------------------------------------------------------------------------
#Variabili PLC **************************************************************************************************************************
tags=['Program:TIM.MISURA_06','Program:TIM.MISURA_OK_06']

#-----------------------------------------------------------------------------------------------------------------------------------------
#Legge le tags nel PLC *******************************************************************************************************************

def leggi_tags():
    value_tags = [] 

    with PLC() as comm:
        comm.IPAddress = '***.***.***.***'
        ret = comm.Read(tags)

    for r in ret:
        #print(r.Value)
        value_tags.append(r.Value)
        
    return value_tags

dati_prima = leggi_tags()
did_change = False
#------------------------------------------------------------------------------------------------------------------------------------------
while True: 
       
    TimeStampAdesso = Data_Ora()  
          
    dati_adesso = leggi_tags()
    print(dati_adesso)
        
    # Loop che comprende tutti i valori e li compara
    items_variati = []  # <- New list
    indici_variati = [] 
    
    for val_adesso, val_prima in zip(dati_adesso, dati_prima):

        # Compara nuovi valori con quelli precedenti
        if val_adesso != val_prima:
            print(f"I valori {val_adesso} {val_prima} sono diversi")

            items_variati.append(val_adesso)  # <- Aggiungi un item alla lista
            #print(items_variati)
                        
            indici_variati.append(dati_adesso.index(val_adesso))
            #print(indici_variati)
            
    did_change = True

    if did_change:
        dati_prima = dati_adesso
        did_change = False
        
    
# ------------------------------------------------------------------------------------------------------------------------------------------
   

# Verifica Misure Analogiche Idrauliche ****************************************************************************************************

# Livello Torre  *************************************************************************************************************************
    
    LivTorreP2 = (dati_adesso[0]) # integer
    LivTorreP2_OK = (dati_adesso[1]) # boolean
    MesLiv = []
    

    if (LivTorreP2_OK) == True :
        MesLiv.append ("Livello "+ str(LivTorreP2/100) + " mt" + " \U0001F603")
        

        if (LivTorreP2/100) > 4.20 and (LivTorreP2_OK) == True:
            MesLiv.append ("Livello "+ str(LivTorreP2/100) + " mt" + " alto \U00002B06") 

        elif (LivTorreP2/100) < 2.30 and (LivTorreP2_OK) == True:
            MesLiv.append ("Livello "+ str(LivTorreP2/100) + " mt" + " basso \U00002B07")

    else:
        MesLiv.append("Livello " + "***" + " mt" + " \U0001F621")
    
#------------------------------------------------------------------------------------------------------------------------------------------
  
    print(TimeStampAdesso)
    print(MesLiv)

    time.sleep(10)

# RaspBerry PI 3
# Raspbian GNU/Linux11 (bulleye)
# Pylogix 0.8.14
# PLC AB CompactLogix

dmroeder, 2023-10-12:
Ah, okay, the details about the network are important.  Seems like maybe the request makes it to the PLC, but you never get a response.  I'll have to recreate the same scenario.

horus61, 2023-10-20:
Good morning everyone. Is there any news regarding this problem? Thanks for your contribution.

dmroeder, 2023-10-20:
Nothing yet.  I'll work on it as soon as I can

dmroeder, 2023-11-08:
I see what is going on @horus61, I'm working on a fix.

dmroeder, 2023-11-09:
@horus61, try 0.9.1

commit 09fe6f24 (2023-11-09): Upped version

Should fix #238