dmroeder/pylogix #247: PLC connection best practice

Issa-Bqain, 2024-05-30:
### When to close connection in a state machine?

Hi, I am looking to pull data continuously from a Rockwell PLC using a finite state machine: 

1) Connect to the PLC and pull data for some tags
2) Serialize data
3) Publish data to a message bus
repeat

I am not entirely familiar with **_how_** exactly the connection to the PLC is made:

1) Should the connection to the PLC be left open indefinitely through all stages of the state machine? what would the impact of this be on the PLC as opposed to constantly opening and closing the connection?

2) Can you recommend any resources I could use to better understand how this library enables connection to the PLC?

I would appreciate some help understanding this, thank you.

kyle-github, 2024-05-30:
There are three stages to a Rockwell PLC connection:

1. TCP connection.  This is usually pretty fast but still on the order of milliseconds.
2. EIP connection.  This is a simple handshake but still takes a request/response pair.
3. CIP connection set up.   You can do disconnected messaging but your maximum packet size is only about 500 bytes.   This takes a while.  Up to 30ms on older PLCs.   With a modern 1756-L8x it is in the low single digit milliseconds.

The whole connection process is time and CPU consuming.   I try to keep connections open as long as I can.

The PLC will drop the connection after some period of idle time (no requests).  I don't remember what Pylogix sets that to.   In my experience it is better to control this from your end of the connection and cleanly close the connection if you are going to have a long idle period.

dmroeder, 2024-05-31:
Pylogix makes the connection to the PLC automatically for you, whenever you call any function (read/write, etc).  When you first create an instance of pylogix.PLC(), a connection is not yet made to the PLC.  When you then call Read(), for example, we'll generate fresh connection parameters (serial number, sequence count), then make a connection to the PLC.  Each time you call Read(), we use the same connection.  If the connection suddenly fails, we'll generate new connection parameters and create a new connection.

If in your code, you end a With statement, the connection will automatically close.  The next time you call Read(), a new connection will be made.

If you create a new instance of pylogix.PLC() in your code, a connection will be made.

As long as you don't close the connection yourself, (calling .Close() or ending your With), the connection does not fail or you wait too long between calling Read/Write/etc. the same connection will be used.

As for when to close the connection, that's a bit up to you.  If I'm reading/writing every minute, I keep the connection open.   If I can't predict when I need to read/write (some external dependency), I'll keep the connection open so I don't end up with some situation where I suddenly get a bunch of requests.  If I'm reading every 10 minutes or every hour, I'll probably close the connection.  I usually error towards keeping the connection open by either reading a tag periodically or requesting the PLC time.

The PLC can handle connections being opened and closed, though I wouldn't make it do extra work unnecessarily.  The 1756-ENBT can be a turd, so definitely don't overburden that module.

Issa-Bqain, 2024-06-07:
@kyle-github @dmroeder Thank you for the help.

Based on above, I would then never close the connection, unless the script terminates (I imagine the connection would drop automatically without calling .Close?). Consider a finite state machine like the one below. If I am reading say 10-20 tags every <1 second, the connection should be left open.

```
from urllib import response
from pylogix import PLC
import requests

#PLC
IP_ADDRESS = "192.168.1.9"
TAG_NAME = "something"
PROCESSOR_SLOT = "2"
IS_MICRO800 = False

#request
API_END_POINT = "127.0.0.1"

# Abstract State class
class State:
    def execute(self):
        pass

# StateA class
class StateA(State):
    def execute(self):
        print("Executing State A")
        # Perform State A specific actions or operations
        # Transition to State B after execution

# StateB class
class StateB(State):
    def execute(self):
        print("Executing State B")
        # Perform State B specific actions or operations
        # Transition to State C after execution

# StateC class
class StateC(State):
    def execute(self):
        print("Executing State C")
        # Perform State C specific actions or operations
        # Transition to State A after execution

class connect_plc(State):

    def execute(self):
        with PLC() as comm:
            comm.ProcessorSlot = PROCESSOR_SLOT
            comm.Micro800 = IS_MICRO800
            comm.IPAddress = IP_ADDRESS
            ret = comm.Read(TAG_NAME)
            print(ret.TagName, ret.Value, ret.Status)

class send_data(State):
    def execute(self):
        
        response=requests.post(API_END_POINT)
        print(response.json())

# State Machine class
class StateMachine:
    def __init__(self):
        self.states = [StateA(), StateB(), StateC()]
        self.current_state = self.states[0]  # Start with State A

    def run(self):
        while True:
            self.current_state.execute()
            self.transition_to_next_state()

    def transition_to_next_state(self):
        current_state_index = self.states.index(self.current_state)
        next_state_index = (current_state_index + 1) % len(self.states)
        self.current_state = self.states[next_state_index]

# Usage
state_machine = StateMachine()
state_machine.run()

```

kyle-github, 2024-06-07:
Make sure you understand when the PLC will close the connection.  I have found that if the PLC closes the connection, it can be messier to clean up and restart another one.  It may depend on a number of factors including the OSes involved.  I always make sure that I (or the underlying library) close the connection cleanly first.

Issa-Bqain, 2024-06-07:
> Make sure you understand when the PLC will close the connection. I have found that if the PLC closes the connection, it can be messier to clean up and restart another one. It may depend on a number [...]

Since I am reading at a time interval which is less than the idle time - shouldn't the PLC _never_ close the connection on its own? At this point, I just close the connection once I terminate the script by choice or if the script/host the script is running on crashes.

kyle-github, 2024-06-07:
The PLC won't unless something else pokes at it.   For instance, if someone puts it in program mode, I think you lose your connections.   Or if someone restarts it.  Etc.

dmroeder, 2024-06-07:
Connections are maintained in program mode.  A PLC reboot will do it, network cable disconnected, too much time between exchanges.

In your case, the PLC won't drop the connection because you are reading frequently.

closed 2024-06-12 by Issa-Bqain