dmroeder/pylogix #250: Connection Issues

Jeffrey2659, 2024-07-15:
### Preflight checks

Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [ X] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [X] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [X ] Have you read previous issues?
- [X ] Is this a supported Rockwell PLC?

## Type of issue

- [ ] Bug
- [ ] Feature Request
- [ ] Question
- [X ] Other

> Delete items that do not apply below.

## Description of issue
 I am not able to read a tag from the PLC, even though all criteria's are seemingly met
## Expected behavior
I am expecting Compressor Running to return  CompressorSelect False Success
## Actual behavior
CompressorSelect None Unknown error timed out
## Code
from pylogix import PLC

# Setup the PLC object with initial parameters
# Change to your plc ip address, and slot, default is 0, shown for clarity
comm = PLC('192.168.10.2', 0)

# Read returns Response class (.TagName, .Value, .Status)
ret = comm.Read('CompressorSelect')
print(ret.TagName, ret.Value, ret.Status)

# Close Open Connection to the PLC
comm.Close()

Please provide a minimal, reproducible example.

https://stackoverflow.com/help/minimal-reproducible-example

## Screenshots
[image]
[image]
[image]

## Stacktrace

## Versions

Include versions to

- pylogix:1.0.1
- plc model: CompactLogix 1769 L30ERM
- python:3.12.4
- OS: Windows

dmroeder, 2024-07-15:
They tester says "PLC is unreachable" and pylogix is not reaching the PLC.

What happens if you PING the PLC?
Post the results of `ipconfig` in the command prompt.

TheFern2, 2024-07-16:
From the screenshot it looks like one Ethernet adapter is on 192.168.0.X

Ping will 100% fail since the tester is using the ping module. You only get that error if the ping fails.

Wondering if you need some routing path in which case tester might not be up to date for routing.

Jeffrey2659, 2024-07-16:
I have attached the screenshot here. I also forgot to mention that the ip addresses studio 5000 is running is on a VM, if that makes a difference. 
[image]
[image]

Jeffrey2659, 2024-07-16:
I am able to ping from VM. I do want to ping from PC, but instead of changing the PC  ip to match the VM, could I change VM ip to match the PC's

TheFern2, 2024-07-16:
What does your vm network settings look like? (Screenshot) also ipconfig from your PC 

If you change the vm to match pc you won't be able to ping the plc anymore.

Jeffrey2659, 2024-07-17:
Got it resolved, but it was probably not the best way. I changed the ip of my vm, and then changed the ip of the plc using a usb. I was able to ping the plc from the usb then. The issue was the host was on a different gateway than the vm and plc as the @TheFern2 mentioned. Thank you guys for helping me diagnose and solve the issue

closed 2024-07-17 by Jeffrey2659

dmroeder, 2024-07-17:
Thanks for the update.  I'm a fan of VM's but they do add an extra layer of complexity.

TheFern2, 2024-07-18:
You can always use another external Ethernet adapter on pc too if you must have pc on 0.X and extra one can be on 10.X

Glad you found a solution.