dmroeder/pylogix #251: PLC read does not run when called from a function

Issa-Bqain, 2024-08-05:
I took the code in the examples and it runs just fine:

### Code from example: 
```
PLC_TAGS_LIST = ['v1','v2']

with PLC() as comm:
    comm.IPAddress = PLC_IP
    ret = comm.Read(PLC_TAGS_LIST)
    for r in ret:
        print(r.Value)
```
### Code from example output (works as expected): 

> some number 1
> some number 2

### Code from example as a function: 

```
PLC_TAGS_LIST = ['v1','v2']

def get_plc_data():
    
  with PLC() as comm:
    comm.IPAddress = PLC_IP
    ret = comm.Read(PLC_TAGS_LIST)
    for r in ret:
        print(r.Value)
```

### Code from example in as a function: 
` `
there is no output

I am wondering if anyone knows why this might be? thanks

Versions: 
- pylogix: 1.0.2
- plc model: Rockwell PLC L8
- python: 3.10.12
- OS: Ubuntu Desktop Linux

dmroeder, 2024-08-05:
Hmm, I'm not seeing this.

```python
with pylogix.PLC("192.168.1.10") as comm:
    print(pylogix.__version__)
    tags = ["BaseDINT", "BaseSTRING"]
    ret = comm.Read(tags)
    for r in ret:
        print(r.Value)
```
Output:
```console
1.0.2
-776185857
XaoOinvQTSuRFlYTPCqerSCIeNXthMjTyAvxQLBwTpTzwKTuUhopoLIKjDrn
```

TheFern2, 2024-08-05:
Are you calling get_plc_data function somewhere? My guess is that you're not.

Issa-Bqain, 2024-08-05:
Hi, Thank you for checking this, I appreciate the help @TheFern2 @dmroeder 
Fern: I had another print statement on each line of the loop to debug which executed and meant the function did run.

This is rather odd but I believe the issue might actually be a networking problem on my end and nothing to do with pylogix. Apologies.

I was accessing a private network (which the PLC is connected to ) remotely through a firewall when I ran the function  `get_plc_data` and it seems that this was causing some issues. I hardwired into the network and everything ran as expected. 

I am still curious as to what caused this response. I just tried it again now and am not able to reproduce this (it ran as expected). 

I will close the issue for now but if I run into this again I will need to dig deeper into the firewall setup as I imagine this is what may have been causing this.

Thank you

closed 2024-08-05 by Issa-Bqain

TheFern2, 2024-08-07:
Would be much valuable to `print(ret)` object since is the Response object, and you should always get it back, it probably would have tell you the exact error, a list of errors is [here](https://github.com/dmroeder/pylogix/blob/master/pylogix/lgx_response.py#L60). If there is a connection issue as you think, then `ret.Value` would be None, and it should still print None afaik, that's just the way python works has nothing to do with pylogix.

Did you have a print statement inside the `with`? For me still prints None

```python
from pylogix import PLC
from pylogix import __version__

with PLC("192.168.1.10") as comm:
    print(__version__)
    tags = ["BaseDINT", "BaseSTRING"]
    ret = comm.Read(tags)
    for r in ret:
        print(r.Value)
```

```
1.0.2
None
None

# if I print the response object
BaseDINT None Register session failed
BaseSTRING None Register session failed
```

Issa-Bqain, 2024-08-07:
> Would be much valuable to `print(ret)` object since is the Response object, and you should always get it back, it probably would have tell you the exact error, a list of errors is [...]

Good note.

I did have multiple print statements to indicate entering the loop but nothing specifically to display `ret.Value` or the like, this is useful for future reference. Thank you.