dmroeder/pylogix #252: documentation on the CIP protocol

kimdiego2098, 2024-08-09:
Hello, can you provide some documentation on the CIP protocol? I need to understand the message format and try to migrate it to other programming languages. Thank you very much

Nekew, 2024-08-09:
Are you asking this repository to teach you CIP?

dmroeder, 2024-08-10:
You have to be an ODVA member in order to get the protocol documentation.  Otherwise, you have to do what many of us did and get friendly with Wireshark

kyle-github, 2024-08-11:
Note that there are already many libraries out there that "talk" CIP.  You can find Go, Rust, C, C++, Python (obviously), Java and more just searching on GitHub.

One large challenge is that the CIP protocol documented by the OVDA is NOT sufficient to talk to any modern CIP-based PLC.   Those PLCs all use vendor-specific extensions to CIP and usually those are not documented or at least not fully documented.   The CIP documentation is a useful base, but it does not really tell you enough to be usable.  It is like learning the grammar for a language but not how people actually speak it.  

And then there are all the weird little quirks like the treatment of BOOL arrays (stop threatening me with that knife, Dustin!).  Rockwell does things one way, Omron does things another.   This is why BOOL arrays in a Rockwell PLC end up being multiples of 32 bits long.  But not in Omron PLCs.

Or strings.   The usual STRING type used in a Rockwell Logix-class PLC is _NOT_ the CIP string type.  It is actually a struct pre-defined in the system.   So all the type info you get is pretty much useless and you have to match on the specific structure ID.   Yay, Rockwell. If Rockwell ever accidentally (or on purpose) changes the struct ID, almost all the libraries out there will break.

Just because that is not enough, you also need to determine alignment of fields within structures and arrays.   Just Omron does it differently between the CJ and NX/NJ models (the first does alignment on 2-byte boundaries and the second does alignment based on the atomic type size more like C does).

As @dmroeder, said, get very friendly with Wireshark.   And work through a lot of examples.

kimdiego2098, 2024-08-12:
Thanks , So I have to gradually improve it through Wireshark , Headache!

And I will close this issue

closed 2024-08-12 by kimdiego2098

dmroeder, 2024-08-12:
> Thanks , So I have to gradually improve it through Wireshark , Headache!
> 
> And I will close this issue

Tell me about it! I've spent a silly amount of time with Wireshark. Though none of this has been all that intuitive to me.

kimdiego2098, 2024-08-12:
What's even sadder is that I don't have the ControlLogix PLC

dmroeder, 2024-08-12:
> What's even sadder is that I don't have the ControlLogix PLC

CompactLogix is enough if you can get your hands on one.  A Micro800 will get you the majority of the way, though there are some notable differences.

TheFern2, 2024-08-12:
I have not used an emulator for sometime but I use softlogix for all my development projects. I want to say emulator works on localhost so it should still work for wireshark.