dmroeder/pylogix #256: Read Array Size limit

Jeremiah-Berger, 2024-11-19:
### Preflight checks

Before you post an issue, ensure you have tried the minimal examples within the repo, or tried the pylogix-tester.

- [X ] Have you tried the [examples](https://github.com/dmroeder/pylogix/tree/master/examples)?
- [ ] Have you tried [pylogix-tester](https://github.com/TheFern2/pylogix-tester)?
- [X ] Have you read previous issues?
- [ ] Is this a supported Rockwell PLC?

## Type of issue

- [ ] Bug
- [ ] Feature Request
- [X ] Question
- [ ] Other

----------

It appears the limit to reading an array is 65535 due to the read command using an int. I am trying to read 80,000 DINT array.
Is the only work around 2 smaller reads?

Code:

comm.IPAddress = '192.168.1.100'
comm.ProcessorSlot = '1'
comm.SocketTimeout = 10.0
ret = comm.Read(DTest[0], 80000)

Error:
struct.error: ushort format requires 0 <= number <= 65535

dmroeder, 2024-11-19:
Currently that is the only work around, I'll check into this.  Thanks!

dmroeder, 2024-11-19:
Gross

kyle-github, 2024-11-19:
CIP.  The protocol that keeps on giving.   Rockwell used 16-bit fields in weird places and then 32-bit fields _for_ _the_ _same_ _value_ in others.

kyle-github, 2024-11-19:
An example is CIP object instance IDs.   Depending on where you get or set an ID, it is 16 bits or 32 bits.  Which is stupid because 65537 instances really is not that many.

dmroeder, 2024-12-29:
@bullfrog64 if you are able to test the branch bugfix/large-list-read, this should be resolved for you

bullfrog64, 2025-01-02:
Yes, it is working. Thanks!

closed 2025-01-02 by bullfrog64

dmroeder, 2025-01-02:
Thanks for checking.  I'm waiting to confirm one more issue, then I'll release a new version with this change.