dmroeder/pylogix #262: Issue with Reading/Writing BOOL Arrays

slinlong, 2025-09-02:
Dear dmroeder,

First of all, I want to express my sincere gratitude for developing the pylogix library - it has been incredibly helpful for my project!

I've encountered an issue with BOOL array read/write operations since version 0.9.2. For example, when I try to access tags like CaseUDTA[8].Out_F[0] to CaseUDTA[8].Out_F[99], the actual operations are performed on CaseUDTA[0].Out_F[0] to CaseUDTA[0].Out_F[99] instead. This has forced me to stay on version 0.9.2 and prevented me from upgrading to newer versions.

Recently, I've been studying the source code, and after analysis, I believe the issue is likely in the _build_ioi function.

Could you please help confirm this issue? Thank you very much!

dmroeder, 2025-09-02:
I will investigate this...

dmroeder, 2025-09-04:
I have a fix that is working @slinlong.  If you're interested, I can push a branch for you to try.

dmroeder, 2025-09-06:
Try this when you get a chance:
https://github.com/dmroeder/pylogix/tree/bugfix/udt-bool-array